VLSIDA/OpenRAM #239: Question: What is the correct allignment in SKY130? Intermediate Pins are not connected to the grid. Likely user mistake.

c-93, 2024-04-29:
**Describe the bug**
_Some_  pins are not connected to the routing grid. In more detail, sometimes zero, one, or two intermediate pins are not connected.
Could be user error.

**Version**
OpenRAM 1.2.48
OpenLane 2

**To Reproduce**
- Modify LEF to 1000 microns database unit as _[described here](https://github.com/VLSIDA/sky130_sram_macros/commit/c2333394e0b0b9d9d71185678a8d8087715d5e3b)_.  
- Instantiate a macro in openlane2, connect it, run synthesis, and fail at the verification step.
- Macro is not connected to the power grid, yet.

I reproduced this bug using two different macros and locations. Different pins were disconnected. So, I guess no explicit conf file needs to be reproduced.

**Expected behavior**
Successfully connecting all pins to the memory macro.

**Logs**
Notice the pattern. Sometime zero skips, then one or two skips.
```log
[DRT-0418] Term my_openram_macro/din0[0] has no pins on routing grid                                                                                                                                                                                                                                                                            
[DRT-0418] Term my_openram_macro/din0[3] has no pins on routing grid                                                                                                                                                                                                                                                                            
[DRT-0418] Term my_openram_macro/din0[6] has no pins on routing grid                                                                                                                                                                                                                                                                            
[DRT-0418] Term my_openram_macro/din0[8] has no pins on routing grid                                                                                                                                                                                                                                                                            
[DRT-0418] Term my_openram_macro/din0[9] has no pins on routing grid                                                                                                                                                                                                                                                                            
[DRT-0418] Term my_openram_macro/din0[11] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[12] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[14] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[15] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[17] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[18] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[20] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[21] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[23] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[26] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[29] has no pins on routing grid                                                                                                                                                                                                                                                                           
[DRT-0418] Term my_openram_macro/din0[31] has no pins on routing grid       
```

**Additional context**
However, in the OpenRoad GUI in step `44-odb-reportdisconnectedpins` ALL pins are visually connected:
[image]

Notice that there is a difference in `HalfGrid` access points for connected and non-connected pins.
Non-functional:  
[image]

Functional:  
[image]

Questions:
- Does the macro need to be manually aligned to a specification?
- Or is this due to the 2:1 LEF unit scaling?

mguthaus, 2024-04-29:
Your image suggests the LEF and connections are fine. I'm not sure what the issue is in the router.

c-93, 2024-04-29:
Thanks for your quick response.  
Yes, I noticed the connections are not always 100% centered within the pin area. But they look connected.

For example pin 0 does not look "perfect" (still visually looks connected). However, it raises the error.
[image]
Pins 1 & 2 look fine and are connected. Pin 3 looks fine, but is not connected.

I manually changed only the [database microns](https://github.com/VLSIDA/sky130_sram_macros/commit/c2333394e0b0b9d9d71185678a8d8087715d5e3b) in the LEF. I did not modify other values. The LEF and instantiation looks good to me.

I can send the exact positions if this helps.

c-93, 2024-04-29:
Thanks again for your feedback.  
I am still not sure what is really happening.
Another pin seems to be more offset, but does _not_ raise an error.
[image]

I agree: Increasing pin size in GDS2 and LEF will be the resonable next step. 

Just a sidenote: [DFFRAM](https://github.com/AUCOHL/DFFRAM) provides pins on another layer and connects without error.
[image]

mguthaus, 2024-04-29:
I would also confirm with the router people why the error is there. These seem like perfectly good connections.

maliberty, 2024-04-30:
It would be helpful to turn on the routing grid in your images.  

> Instantiate a macro in openlane2, connect it, run synthesis, and fail at the verification step.

Is too vague.  Please make a standalone routing test case that shows the problem.

c-93, 2024-04-30:
Thanks for the suggestion! I will prepare a standalone test, but need some days.  

I activated the grid view in openroad, but I need to double-check if the grid settings correspond to the router grid settings from openlane2.

[image]

c-93, 2024-05-02:
As suspected in the initial issue, I discovered this was a user error. Thanks for the responses so far.

My solution/approach:

- The red macro connections to the macro are on the `metal 4 layer`. I have seen this in the openroad gui.

- The non-modified default settings from the `openlane2's` `openroad-globalrouting.log` are:

[INFO GRT-0088] Layer li1     Track-Pitch = 0.4600  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met1    Track-Pitch = 0.3400  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met2    Track-Pitch = 0.4600  line-2-Via Pitch: 0.3500
[INFO GRT-0088] Layer met3    Track-Pitch = 0.6800  line-2-Via Pitch: 0.6150
[INFO GRT-0088] Layer **met4**    Track-Pitch = **0.9200**  line-2-Via Pitch: 1.0400
[INFO GRT-0088] Layer met5    Track-Pitch = 3.4000  line-2-Via Pitch: 3.1100

- The `Track-pitch` is defined as the minimum distance from track-center to track-center.

- By moving the macro: The center of the leftmost pin is now at X-Position 138,46, which makes `138,46 / 0,92=150,5`, matching the grid. The error is solved.

If my explanation is correct:  
Maybe we should add this alignment explanation to the openlane2 and/or the OpenRAM documentation?

c-93, 2024-05-06:
Moving the macro to the correct pin coordinate works only if the pins on the upper and lower sides both match to the routing grid. However, this does not always seem to be the case. Thus, the lower side io pins connect, but some upper side pins raise an error - as they mismatch the grid.

> Yes, making the pins a bit bigger might ensure that all pins align in the center and the router doesn't give this error.

I am not sure which place is correct for implementing this modification.
As the `add_io_pin` [function](https://github.com/VLSIDA/OpenRAM/blob/b6a6f12642df6b84facc24a77f9a6f67a0d62dab/compiler/base/hierarchy_layout.py#L1904) instantiates a `power_pin`, [this looks like the correct place](https://github.com/VLSIDA/OpenRAM/blob/b6a6f12642df6b84facc24a77f9a6f67a0d62dab/compiler/base/hierarchy_layout.py#L1919).

However, modifying it for example from
`min_area = drc["minarea_{}".format(self.pwr_grid_layers[1])]`
to
`min_area = drc["minarea_{}".format(self.pwr_grid_layers[1])]` `*2`
only scales all internally existing IO pins (marked in red), but does not scale the desired pins that actually connect the macro to the surrounding world (marked in blue).

[image]

How can I scale the "outer-world-pins"?

mguthaus, 2026-10-16:
Closing as stale. No activity for about 2.4 years and no reproduction against current `dev`, so there is nothing actionable here as it stands.

This was a question rather than a defect, and it was answered in the thread. If there is a concrete alignment bug left over, a fresh issue with the configuration and the offending geometry would be easier to act on than this thread.

To be clear this is closed for inactivity, not rejected. If it still affects you please reopen or file fresh, with a reproduction against current `dev` including your configuration file, technology, OS and `python3 --version`, and the temp-directory contents OpenRAM leaves on failure ([the FAQ lists which files](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself a PR against `dev` would be welcome and we will review it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md).

closed 2026-10-16 by mguthaus