VLSIDA/OpenRAM #242: Power estimation seems too high

Lukemagik, 2024-05-24:
**Describe the bug**
I ran the sky130_sram_1rw_tiny example and the power report shows hundreds of watts, which seems wrong. Could this be a bug? How can I solve it?
[image]

**Version**
1.2.48

**To Reproduce**
cd OpenRAM/macros
make sky130_sram_1rw_tiny

mguthaus, 2024-05-24:
It is a known issue that the power numbers have not been calibrated. The cell powers in the tech file are not correct.

Lukemagik, 2024-05-24:
Hi, thank you for the response. Is it possible to obtain the power estimation in another way?

mguthaus, 2024-05-24:
You can run a spice simulation. 

These are the placeholder values for leakage:
https://github.com/VLSIDA/OpenRAM/blob/stable/technology%2Fsky130%2Ftech%2Ftech.py#L744

Similarly, there are estimates for capacitance used in dynamic power which might be set correctly.

Lukemagik, 2024-05-24:
Thank you very much. With this approach, is it possible to obtain the .lib file as well? I would like to use these memories as macros in another design on OpenLane and I would like to have a complete power estimation.

Lukemagik, 2024-05-24:
I didn't quite understand this last part.
In this [case ](https://github.com/VLSIDA/sky130_sram_macros/tree/main) how was the power estimated on ? It seems to have more accurate values?

Lukemagik, 2024-05-24:
Sorry @mguthaus , in this [issue ](https://github.com/The-OpenROAD-Project/OpenLane/issues/1838)it seems to me that you have the same problem as mine. I have a design on OpenLane that uses OpenRAM macros and I need to estimate the entire system power. However, to do this, it seems from this issue that a gate-level netlist is needed rather than the .lib file. Were you able to solve this problem?

mguthaus, 2024-05-24:
There are no gate-level netlists of SRAM macros. It must use a .lib file.

Lukemagik, 2024-05-27:
I understand, but with OpenLane, without the gate-level netlist of a macro, I need to use ///sta-blackbox, which seems to ignore the .lib file.

My question is, with the .lib file but without the gate-level netlist, is it possible to estimate the power of these memories within another design, and if so, how?

Lukemagik, 2024-05-27:
I understand, however, it seems that OpenLane is unable to read the .lib files of the macros to calculate power, and it only works when the gate-level netlist is present.

So my question is, based on this [issue](https://github.com/The-OpenROAD-Project/OpenLane/issues/1838), how can we include the memory macro in power calculation when we only have the .lib file available?

mguthaus, 2024-05-27:
This would be an openlane issue. Gate level doesn't exist for any memory macro even with other tools. It doesn't use the cells from the library.

Openlane should be using the .lib file for the memory macro to get the power.

Lukemagik, 2024-05-27:
Perfect, so you're confirming that it's impossible for OpenLane to estimate the power of these memories since only the .lib file is available?

mguthaus, 2024-05-27:
I'm confirming that openlane should only need the lib file to estimate power.

Lukemagik, 2024-05-27:
So, did you manage to solve this [issue](https://github.com/The-OpenROAD-Project/OpenLane/issues/1838)?

mguthaus, 2026-10-16:
Closing as stale. No activity for about 2.4 years and no reproduction against current `dev`, so there is nothing actionable here as it stands.

Partly superseded: #294 reports suspect `internal_power` values in generated sky130 Liberty files with current detail and is still open, which is probably the better place for this. Worth knowing that #313 fixed `isapproxdiff`, which had been comparing each Liberty golden against itself, so the power numbers in `compiler/tests/golden/` are less validated than their passing status implied.

To be clear this is closed for inactivity, not rejected. If it still affects you please reopen or file fresh, with a reproduction against current `dev` including your configuration file, technology, OS and `python3 --version`, and the temp-directory contents OpenRAM leaves on failure ([the FAQ lists which files](https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/FAQ.md)).

If you would like to fix it yourself a PR against `dev` would be welcome and we will review it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md).

closed 2026-10-16 by mguthaus