VLSIDA/OpenRAM #245: Runtime

mbautista-lab, 2024-06-25:
Hi I am trying to run a 16kbyte SRAM on sky130.  Is it normal that it will take couple of weeks to generate? 
Mine has been running for 1 month now.  It generate a temp.gds and I don't see any errors flag but its still finishing. 
So Im curious to know  is couple of weeks a normal runtime for ex. 16kbyte memory ?

mguthaus, 2024-06-25:
What step is running for so long? This is the largest memory block that I've generated.

Likely, I suspect it is timing characterization. This can be sped up a bit if you have parallel simulation tools, but it's ultimately a bottleneck that needs to be addressed. 

The alternative now is to rely on the regression models, but these are not calibrated. We would welcome contributions to help with this

mbautista-lab, 2024-06-25:
It is still running please see the below image. 
[image]

When you say a parallel simulator, can we use xyce ? 

Thanks

mguthaus, 2024-06-25:
Yes, Xyce would be a parallel simulator, but it needs to be compiled as such.

This is stuck running extraction. Which version of magic are you using? There is likely an issue with that.

mbautista-lab, 2024-06-25:
My magic is version 8.3.363

What should be the correct version to use?

mguthaus, 2024-06-25:
That is correct:
https://github.com/VLSIDA/OpenRAM/blob/0937f86761e727ae3ba6c3e0d6df8a84a760cf8a/install_conda.sh#L10

What kind of machine? How much memory is it using?

mbautista-lab, 2024-06-26:
Im using linux server, about 1.75T memory

mguthaus, 2024-06-26:
It's using 1.75TB?? That is the problem, it is paging to swap. This sounds suspicious though. 

How much physical memory is in the machine?

mguthaus, 2024-06-26:
It could be a memory leak in magic.

TheAnimatrix, 2026-10-13:
Root cause of the extraction stall, with numbers. Setup: sky130 1rw1r 2048x32 (65,536 bitcells, words_per_row 8), OpenRAM 1.2.48 plus the #320 patch, Magic 8.3.363 from install_conda.sh, 24 cores, 43 GB, no swap. Same symptom as this issue: run_ext sits on the bitcell_array for hours at one core, RSS flat around 4 GB, so it is not paging.

**1. OpenRAM puts 2.5 labels per bitcell on the array cell.** `bitcell_base_array.route_supplies()` copies every instance's vdd and gnd pin up to the array. The array .mag for this macro has 165,632 labels on 65,536 cells, all sticky: 98,560 gnd, 65,536 vdd, 1,536 bl/br/wl. Plus 132,865 pin rects. A 640x16 array (10,240 cells) has 26,240.

**2. `extract unique all` is quadratic in those labels.** perf on the live process: 87% in `extMakeUnique` + `DBRemoveLabel` (ExtUnique.c:285, DBlabel.c:498). Every duplicate vdd/gnd gets renamed, and each rename rescans the cell's label list. Magic 8.3.685 spends about 32 min in this pass on the array cell alone; 8.3.363 is slower.

**3. 8.3.363 also scans the whole sticky-label list for every tile of every subcell during hierarchical extraction** (`extHierConnectFunc1`, ExtHier.c:395 is the hot line, 23% self time even with the vdd/gnd labels stripped). Magic commit 1d8fcca0 (2023-04-06, "Limiting this processing significantly cuts down on processing time when there are many labels in a layout") is the fix. 8.3.363 is dated 2023-01-17, eleven weeks earlier. 8.3.363 additionally segfaults in `extract unique all` when this hierarchy is loaded from .mag files.

Same array cell, `extract unique all` + `extract all`:

| Magic | labels as generated | vdd/gnd labels stripped |
|---|---|---|
| 8.3.685 | 36 min | 2 min 32 s |
| 8.3.363 | killed after 4.5 h, 0 merge lines written | still running |

Whole macro on 8.3.685: extraction + ext2spice 48:57, 4.8 GB peak RSS, 1,309,696 merge lines in the array .ext.

Fixes, in order:
- OpenRAM: stop copying per-instance supply pins in `route_supplies`. One pin shape per rail is enough for the parent to connect. Removes both costs on any Magic version.
- Drop the 8.3.363 pin in install_conda.sh. The nix flake already pulls current Magic.
- Magic: `extMakeUnique` could avoid the linear `DBRemoveLabel` scan, but the first fix makes that moot for OpenRAM.

Separate finding while here: `run_drc.sh` feeds Magic on stdin and the error count is printed before the background check has run. An injected violation still reports `Total DRC errors found: 0`; the same commands from a script file report 7 error tiles on this layout and 10 on the 640x16. Will file separately.

TheAnimatrix, 2026-10-13:
Follow-up with the missing numbers, now from a symbol build of 8.3.363 so the hot lines are exact.

Same array cell, `extract unique all` + `extract all`:

| Magic | labels as generated | vdd/gnd labels stripped |
|---|---|---|
| 8.3.685 | 36 min | 2 min 32 s |
| 8.3.363 | unique pass 19 min, then 67 min in the array with 0 merge lines, killed | 3 min 46 s |

So the labels are the whole cost. Even 8.3.363 finishes the 65,536-cell array in under 4 min once the per-instance vdd/gnd labels are gone. perf on 8.3.363 during the array stall: 66% of samples on ExtHier.c:395 and :397, the loop in `extHierConnectFunc1` that walks the sticky-label list for every tile of every subcell. All 165,632 array labels are sticky. That is the loop commit 1d8fcca0 (2023-04-06) limits.

Next wall after extraction is netgen. `sky130_fd_bd_sram__openram_dp_cell` never matches at cell level: the extracted cell has 16 devices against 10 in the .sp, the extra ones being edge transistors whose far terminal is a cell-boundary port (`a_38_n79#`, `a_400_n79#`). After flattening dp_cell, the array itself still mismatches by 1,024 devices (4 per column, 656,896 vs 655,872) and 1,279 nets, so netgen flattens the whole array into the bank. On 10,240 cells that costs 8.5 min. On 65,536 cells netgen ran 4 h 25 min, reached 10 GB, and was killed by the memory cap while flattening the array. A second run under a 24 GB cap is in progress. Modelling the edge devices so dp_cell and the array match at their own level would avoid the flatten entirely.

mguthaus, 2026-10-13:
Happy to accept PRs. Can you verify that LVS still works with the labels removed in all technologies supported? Often these labels were added to improve candidate net matching in LVS but may have been too liberally added.

mguthaus, 2026-10-13:
And I am well aware of the parasitic devices that span adjacent bitcells, but we don't design the bitcell -- it was Skywater. If you have a way around this, let me know, but otherwise the correct behavior is to flatten the bitcell and match at the array level.

mguthaus, 2026-10-13:
Once you get to 16kb, it is more likely that you will need multiple banks rather than one single bank.