VLSIDA/OpenRAM #246: Adding and Removing Modules and Customizing Module Ports

YWJ226, 2024-07-03:
**Describe the bug**
Hello, I encountered some issues while using OpenRAM and I'm looking forward to your assistance.

1、
Does OpenRAM support customizing the number of ports for any module? For example, when I want to change the five ports of the decoder (input, input, output, power, ground) to three ports (input, input, output), the following error occurs.
[image]

2、
Does OpenRAM support adding and deleting modules? For example, I want to add a custom module between the decoder and the bitcell, and I also need to delete the wordline_driver module.

**Version**
1.2.48

mguthaus, 2024-07-03:
Hi,

In short, these would require you to modify the next level up in hierarchy with a custom module.

Hope that helps.

YWJ226, 2024-07-03:
Thank you for your response. May I confirm if the only required actions are making the following changes in the `tech.py` file.
[image]

 And making the corresponding adjustments to the relevant modules in the `custom` folder of my technology library?

mguthaus, 2024-07-03:
I believe that would solve your pin mismatch issue. But you would also need to ensure that the supply pins are actually connected properly as well.

closed 2024-07-03 by mguthaus