VLSIDA/OpenRAM #247: Error in using sram macro(tiny) in openlane2

FriedrichWu, 2024-07-19:
**Describe the bug**
Hello,

I encountered some errors in trying to use sram macros in openlane2. The openlane2(magic) complains that some cells are placed at itself, and unknown layer/datatype in boundary. I also try just to use only the bank extracted from the sram macro, but failed with similar error. 

[image]

I'm not sure if it's about the gds file, but it looks fine when opening with klayout

There is also one similar error -> https://github.com/The-OpenROAD-Project/OpenLane/issues/1040

**Version**
Openram1.2.48,  Openlane 2.0.0rc2(but 2.04 has same problem)

**To Reproduce**

[macro_test.zip](https://github.com/user-attachments/files/16310272/macro_test.zip)

**Expected behavior**
No error in reading gds

**Logs**
Logs contained in the uploaded file

mguthaus, 2024-07-19:
> There is also one similar error -> The-OpenROAD-Project/OpenLane#1040

The layer errors are complaining about necessary layers in the macro. These
are for OPC etc and cannot be removed.

FriedrichWu, 2024-07-20:
Thanks for the quick response.

I see. Then it's probably the problem of the openlane. Is there maybe any workaround to make the tool to recognize those necessary layers?

mguthaus, 2024-07-20:
It's shown as an error, but it is really a warning.

mguthaus, 2026-10-13:
Still open. This may be partly on the OpenLane side rather than OpenRAM's, since it concerns consuming the generated macro rather than producing it.

One thing that has changed since: #322 just fixed the sky130 LEF header on `dev`, which declared `DATABASE MICRONS 2000` when sky130 tech LEFs use 1000 (#278). That broke macro import in OpenROAD-based flows, so it is worth retesting against `dev` before digging further.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.