VLSIDA/OpenRAM #248: Generating SRAM gets stuck in extraction process of magic

FriedrichWu, 2024-07-30:
**Describe the bug**
Hello,

I'm trying to generating 2kByte Sram, but the process gets stuck in run_ext for nearly one day. This issue seems to occur more when sram size is large. I could make the whole sram generation process be able to finish by disabling lvs & commenting the stimulus and measurement part, but this is of course not a very good solution.

This is a similar issue like here->https://github.com/VLSIDA/OpenRAM/issues/245

And in the .ext.err file I find a lot of warnings, besides "unknown layers" and "boundary was redefined", there is also error shows "Error:  Asymmetric device with multiple terminals!", which is also mentioned here->https://web.open-source-silicon.dev/t/424007/i-m-trying-the-sky130-support-in-openram-using-the-example-c 

**Version**
The current latest version

**Expected behavior**
The process could finish without error

**Logs**
[myconfig_sky.ext.err.zip](https://github.com/user-attachments/files/16424552/myconfig_sky.ext.err.zip)

mguthaus, 2024-07-30:
This sounds like a run time bug with magic. There isn't much we can do to speed up the extraction on the OpenRAM side. 

My only questions are how much memory you have and are using. Especially if you are running it in Docker or a VM.

mguthaus, 2024-07-30:
The asymmetric device message is expected since the SRAM cells have an asymmetric transistor or two.

hwsyydw, 2025-07-07:
> > This sounds like a run time bug with magic. There isn't much we can do to speed up the extraction on the OpenRAM side.这听起来像是 magic 的运行时错误。我们无能为力来加快 OpenRAM 端的提取速度。
> > My only questions are how much memory you have and are using. Especially if you are running it in Docker or a VM.我唯一的问题是你有多少内存和正在使用多少内存。特别是如果您在 Docker 或 VM 中运行它。
> 
> Thanks for your quick reply. I'm running on Linux sever, it has around 124 GB memory available and another 37 GB for swap感谢您的快速回复。我在 Linux 服务器上运行，它有大约 124 GB 的可用内存和另外 37 GB 的交换内存

Please how did you solve it in the end, I also ran into the problem in the running example

mguthaus, 2026-10-13:
This should be substantially improved on `dev` by #320 (merged as de5614a9). Please give it a try and reopen if you are still stuck.

The Magic script that extracts the layout netlist for LVS was running `extract all` with capacitance, coupling and resistance extraction enabled, and then `ext2spice` threw all of it away (`cthresh infinite`, `rthresh infinite`) before netgen ever saw it. On large arrays that extraction was most of the runtime, which is what looks like a hang in `run_ext`. The fix disables those three extractor options for the LVS netlist only; DRC runs in a separate Magic process and PEX runs its own extraction with the defaults, so neither is affected.

Measured on scn4m_subm here, with DRC and LVS both run and verdicts unchanged (0 DRC errors, netgen matching uniquely, identical device and net counts):

| design | extract before | after | `.ext` before | after |
|---|---|---|---|---|
| 128 words x 16 | 16.0 s | 11.0 s | 4.55 MB | 2.00 MB |
| 256 words x 16 | 28.0 s | 18.0 s | 7.54 MB | 3.17 MB |

The saving grows with array size, and the contributor reported their 2 KB sky130 macro going from an overnight timeout to about 2 minutes, which matches your case closely.

I am closing this as addressed rather than conclusively fixed, since I could not reproduce your original 2 KB sky130 run here — the sky130 cell library is not installed on my machine. If it is still slow on `dev`, please reopen with the new timings and I will look again.

On the `.ext.err` warnings you mentioned: "unknown layers", "boundary was redefined" and "Asymmetric device with multiple terminals" are emitted identically before and after, so they are unrelated to the runtime. If any of them turn out to matter they are worth a separate issue.

closed 2026-10-13 by mguthaus