VLSIDA/OpenRAM #249: Modiftying Openram

Redroom20, 2024-08-13:
Hello, can anyone please help me figure out the flow of openram that when i run openram what is the driver code, where decoder and arrays are formed as we are thinking of replacing sram with some other kind of ram for that i need to figure out these things.
If anyone can tell it'll be very helpful

mguthaus, 2024-08-13:
Hi,

This is not really an issue, so I'm going to close it. 

If you want help, the best place is the OpenRAM slack workspace linked in the documentation. I will answer questions there. 

A quick response: The top level now depends on whether you are creating a ROM, RAM, or characterizing.  For SRAMs, there are 3 "top levels":
sram_compile.py - create an SRAM
sram_char.py - characterize an SRAM
sram_func.py - functionally simulate (in Spice) an SRAM

closed 2024-08-13 by mguthaus

Redroom20, 2024-08-13:
okay, thanks for your response