VLSIDA/OpenRAM #251: Assertion Error in SRAM Generation with Specific word_size and num_words Configurations

riagus99, 2024-08-18:
**Describe the bug**
I attempted to generate an SRAM instance using the following command:
```
python $OPENRAM_HOME/../sram_compiler.py myconfig.py
```
While some configurations work perfectly, others fail to execute. I'm not sure if I'm using an unsupported configuration or if this could be a bug in OpenRAM.

Below, I've listed the configurations that work and those that do not.

Working configurations
- word_size = 8, num_words = 32
- word_size = 64, num_words = 2048

Not working configurations
- word_size = 32, num_words = 100
- word_size = 48, num_words = 96
- word_size = 96, num_words = 96

**Version**
commit [b6a6f12](https://github.com/VLSIDA/OpenRAM/commit/b6a6f12642df6b84facc24a77f9a6f67a0d62dab)

**To Reproduce**
As an example, here is the configuration that fails (word_size = 32, num_words = 100).

python version: Python 3.8.19

configuration file (`myconfig.py`)
```
num_rw_ports    = 1
num_r_ports     = 0
num_w_ports     = 0

# Number of bits per word
word_size = 32
# Number of words (some people call this 'depth')
num_words = 100

# The fabrication technology. This must match the PDK name in $OPENRAM_TECH.
tech_name = "freepdk45"

# Process corners, temperature and voltage to characterize
process_corners = ["TT"]
supply_voltages = [0.9]
temperatures = [40]
nominal_corner_only = True

route_supplies  = False
#check_lvsdrc = True

#output_path     = "SRAM_{0}x{1}_r1_w1".format(word_size, num_words)
#output_name     = "SRAM_{0}x{1}_r1_w1".format(word_size, num_words)
#instance_name   = "SRAM_{0}x{1}_r1_w1".format(word_size, num_words)
output_path     = "SRAM_{0}x{1}_rw{2}_r{3}_w{4}".format(word_size, num_words, num_rw_ports, num_r_ports, num_w_ports)
output_name     = "SRAM_{0}x{1}_rw{2}_r{3}_w{4}".format(word_size, num_words, num_rw_ports, num_r_ports, num_w_ports)
instance_name     = "SRAM_{0}x{1}_rw{2}_r{3}_w{4}".format(word_size, num_words, num_rw_ports, num_r_ports, num_w_ports)
```

**Logs**
Below is the output I received:
```
|==============================================================================|
|=========                      OpenRAM v1.2.48                       =========|
|=========                                                            =========|
|=========               VLSI Design and Automation Lab               =========|
|=========        Computer Science and Engineering Department         =========|
|=========            University of California Santa Cruz             =========|
|=========                                                            =========|
|=========          Usage help: openram-user-group@ucsc.edu           =========|
|=========        Development help: openram-dev-group@ucsc.edu        =========|
|=========                See LICENSE for license info                =========|
|==============================================================================|
** Start: 08/18/2024 07:42:04
Technology: freepdk45
Total size: 3200 bits
Word size: 32
Words: 100                                                                                                                                                                                                                                                                                                                                                           Banks: 1                                                                                                                                                                                                                                                                                                                                                             RW ports: 1                                                                                                                                                                                                                                                                                                                                                          R-only ports: 0                                                                                                                                                                                                                                                                                                                                                      W-only ports: 0                                                                                                                                                                                                                                                                                                                                                      Design supply routing skipped. Supplies will have multiple must-connect pins. (route_supplies=True to enable supply routing).                                                                                                                                                                                                                                        DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
DRC/LVS/PEX is disabled (check_lvsdrc=True to enable).                                                                                                                                                                                                                                                                                                               Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).                                                                                                                                                                                                                                                                   Only generating nominal corner timing.                                                                                                                                                                                                                                                                                                                               Words per row: None                                                                                                                                                                                                                                                                                                                                                  Output files are:                                                                                                                                                                                                                                                                                                                                                    /root/verilog_works/OpenRAM/SRAM_32x100_rw1_r0_w0/SRAM_32x100_rw1_r0_w0.lvs                                                                                                                                                                                                                                                                                          /root/verilog_works/OpenRAM/SRAM_32x100_rw1_r0_w0/SRAM_32x100_rw1_r0_w0.sp
/root/verilog_works/OpenRAM/SRAM_32x100_rw1_r0_w0/SRAM_32x100_rw1_r0_w0.v                                                                                                                                                                                                                                                                                            /root/verilog_works/OpenRAM/SRAM_32x100_rw1_r0_w0/SRAM_32x100_rw1_r0_w0.lib                                                                                                                                                                                                                                                                                          /root/verilog_works/OpenRAM/SRAM_32x100_rw1_r0_w0/SRAM_32x100_rw1_r0_w0.py                                                                                                                                                                                                                                                                                           /root [... cut ...]                                                                                                                                                                                                                                                   
```

**Additional context**
I have reviewed the documentation but could not find any constraints related to the configuration that could explain this issue. If there are specific constraints that I should be aware of, could you please provide some guidance?

mguthaus, 2024-08-18:
Yes, this is a duplicate of https://github.com/VLSIDA/OpenRAM/issues/228

srd8668, 2025-04-21:
Thanks for the reply i will try this and let you know if i face any other problem.

srd8668, 2025-04-30:
Hi,
I was able to generate SRAM with the below config, my question here is currently i made analytical_delay=True and generated, can i use the generated gds and lef files fabrication or should i generate it by making analytical_delay=False and use it for my Chip. Iam intending to have a dual port sram for parallel read and write operations with operational frequency of 200MHZ.

# Data word size
word_size = 32
# Number of words in the memory
num_words = 256
## need to check what the below does 
write_size = 8

num_rw_ports = 2

# Technology to use in $OPENRAM_TECH
tech_name = "sky130"
# Process corners to characterize
process_corners = [ "TT" ]
# Voltage corners to characterize
supply_voltages = [ 1.8 ]
# Temperature corners to characterize
temperatures = [ 100 ]

# Output directory for the results
output_path = "temp_32_1024_master_last_stable_2rw"
# Output file base name
output_name = "sram_32x1024"
check_lvsdrc=True
output_datasheet_info = True
netlist_only = False

# Disable analytical models for full characterization (WARNING: slow!)
analytical_delay = True

# To force this to use magic and netgen for DRC/LVS/PEX
# Could be calibre for FreePDK45
drc_name = "magic"
lvs_name = "netgen"
pex_name = "magic"

mguthaus, 2025-04-30:
I would suggest that you get simulation working to verify a 200MHz operation.

srd8668, 2025-04-30:
Simulation in the sense of SPICE simulation, or are you referring to RTL simulation with the Verilog model which got generated to verify the 200 MHz operation?

mguthaus, 2025-04-30:
Spice.

mguthaus, 2025-05-03:
The characterization model used won't change the layout and implementation. So both will result in the same design layout, but the .lib file with delay and power will be different.

srd8668, 2025-05-03:
Thanks for the information.

mguthaus, 2026-10-13:
Status update: there is a fix in review for this as #321.

I reproduced your failure on current `dev`. `lib.py` and `sram.save()` characterize at the address with every bit set, which is only a real word when `num_words` is a power of 2. For 96 words with `words_per_row = 2` the address is 5 bits short of the mark — all ones is 127 while the last word is 95 — and characterization stops with `Could not find bl net in timing paths`. Same for your 100-word case (probe 127, last word 99).

#321 replaces the all-ones probe with the last valid word. I verified the arithmetic across both of your sizes and several others, and power-of-2 configurations are unchanged bit-for-bit, so existing results do not move.

I have asked the contributor for one change to the accompanying test before merging. I will update here when it lands.