VLSIDA/OpenRAM #261: DRC Issues with L-shaped Poly and Different Width/Spacing Requirements when Porting OpenRAM to a New Technology Library

TemLos, 2025-02-19:
**Describe the bug**
A clear and concise description of what the bug is.

I am currently working on porting OpenRAM to a new technology library and have encountered several DRC issues. Specifically, some of the generated cells use L-shaped poly, which is not supported by the new technology library I am working with. Additionally, the new library has different width and spacing requirements for vertical and horizontal poly, which further complicates the situation.
I was wondering if anyone has encountered similar issues or could provide suggestions on how to address these problems. Any insights or recommendations would be greatly appreciated!

**Version**
Which commit are you using?
1.2.48

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

**Expected behavior**
A clear and concise description of what you expected to happen.
If possible, I need a solution that can help me address these DRC issues.

**Logs**
If applicable, add logs or output to help explain your problem.

**Additional context**
Add any other context about the problem here.

mguthaus, 2025-02-19:
Do you mean poly to connect multiple fingers?

No, we haven't seen this so it would need custom implementations of the base generator gates. It would be a significant rewrite. I'm not sure of any other way.

TemLos, 2025-02-19:
> Do you mean poly to connect multiple fingers?
> 
> No, we haven't seen this so it would need custom implementations of the base generator gates. It would be a significant rewrite. I'm not sure of any other way.

Understood, thank you very much for your help.

closed 2025-02-19 by TemLos