VLSIDA/OpenRAM #262: Documentation clarifications in basic_usage and basic_setup

sampollard, 2025-02-26:
Hi everyone. I'm starting to work with OpenRAM and I think it's a great project. I had a few confusions I'd like to post here in case others got stuck like I did.

I think the directory names have changed, I needed to set `OPENRAM_HOME` to `export OPENRAM_HOME="path/to/OpenRAM/compiler"` (the capitalization is off, the info is correct in `basic_setup.md` however)

Moreover, it wasn't immediately clear to me how to get the python environment working - I think miniconda/anaconda caused more problems than it fixed. Might I suggest also recommending using `venv` in `basic_setup.md` as follows?

```
cd OpenRAM
python3 -m venv .
source bin/activate
python3 -m pip install openram
```

(as a MacOS/homebrew user, trying to do anything with the local `pip3` command ends up being more annoying and I've had better luck with `venv` versus Anaconda or the system `pip`).

Another small documentation clarification: to pass arguments into the macros Makefile, one needs to pass, e.g., `make OPENRAM_OPTS=-n example_config_scn4m_subm`.

mguthaus, 2025-02-26:
Hi Sam,

Thanks for the suggestions. If you could make specific commit changes, I can merge them in. However, nobody is really funded to maintain openram at the moment, so this will like sit for a while.

mguthaus, 2026-10-13:
Still open, and a documentation PR is the quickest path. `docs/source/basic_usage.md` and `basic_setup.md` are both plain Markdown, so the clarifications you listed can go straight in.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.

mguthaus, 2026-10-27:
Addressed on `dev` by #332, which covers all three points you raised:

- **`OPENRAM_HOME` capitalization** — fixed in `basic_usage.md` to match the actual repository name.
- **Python environment** — `basic_setup.md` now has a Python Environment section recommending `venv` plus `pip install -e .`, instead of leaving conda as the implied route. I verified the editable install works and that `import openram` / `sram_compiler.py -h` then succeed from an unrelated directory.
- **`make OPENRAM_OPTS=…`** — now documented for the macros Makefile. One wrinkle worth knowing: a command-line assignment takes precedence over the Makefile's `OPENRAM_OPTS += -v --keeptemp`, so `OPENRAM_OPTS=-n` also drops those defaults.

While checking the patch I confirmed a related inaccuracy you were close to: the "add `compiler` to `PYTHONPATH`" bullet in `basic_usage.md` does not actually make `import openram` work, because the package is the repository root rather than `compiler/`. The new install instructions are the route that works. Tracking that separately.

Thanks for taking the time to write up where you got stuck — reports like this are what make the docs better for the next person. Closing as fixed.

closed 2026-10-27 by mguthaus

sampollard, 2026-10-27:
Thanks for doing this - sorry I wasn't able to make the MR myself. I'm glad the project is active again! I can also now share the use case at Sandia for OpenRAM since it's open source, it's called [circulette](https://github.com/sandialabs/circulette). There's features of Verilog you use that we don't support but we do intend on evaluating our approach on OpenRAM in the future.