VLSIDA/OpenRAM #266: Can openram add writemask port in this version?

cinder023, 2025-03-24:
sram22_1024x32m8w32 mem_0_0 (
    .addr(mem_0_0_addr),
    .clk(mem_0_0_clk),
    .din(mem_0_0_din),
    .dout(mem_0_0_dout),
    .we(mem_0_0_we),
    .wmask(mem_0_0_wmask)
  );
As it shows.In my design ,there is a write mask port in sram,can Openram add the writemask port in the config?
If can,what should i do?

mguthaus, 2025-03-24:
Yes, you can specify both the word_size and write_size in the config file. This will give you a wmask bit for every word_size/write_size. They must be multiples.

closed 2025-03-24 by mguthaus

cinder023, 2025-03-24:
> Yes, you can specify both the word_size and write_size in the config file. This will give you a wmask bit for every word_size/write_size. They must be multiples.

Thanks for your help
By the way,i want to generate the wmask port in sram.v ,which keywords should i add in the config.py
There is no writemask port in the sram generated by OpenRam in default config
module sram_8_256_freepdk45(
`ifdef USE_POWER_PINS
    vdd,
    gnd,
`endif
// Port 0: RW
    clk0,csb0,web0,addr0,din0,dout0
  );

  parameter DATA_WIDTH = 8 ;
  parameter ADDR_WIDTH = 8 ;
  parameter RAM_DEPTH = 1 << ADDR_WIDTH;
  // FIXME: This delay is arbitrary.
  parameter DELAY = 3 ;
  parameter VERBOSE = 1 ; //Set to 0 to only display warnings
  parameter T_HOLD = 1 ; //Delay to hold dout value after posedge. Value is arbitrary

`ifdef USE_POWER_PINS
    inout vdd;
    inout gnd;
`endif
  input  clk0; // clock chipyard(*)
  input   csb0; // active low chip select chipyard(x)
  input  web0; // active low write control	chipyard(*)
  input [ADDR_WIDTH-1:0]  addr0;		//chipyard(*)
  input [DATA_WIDTH-1:0]  din0;		//chipyard(*)
  output [DATA_WIDTH-1:0] dout0;		//chipyard(*)

  reg [DATA_WIDTH-1:0]    mem [0:RAM_DEPTH-1];

  reg  csb0_reg;
  reg  web0_reg;
  reg [ADDR_WIDTH-1:0]  addr0_reg;
  reg [DATA_WIDTH-1:0]  din0_reg;
  reg [DATA_WIDTH-1:0]  dout0;

mguthaus, 2025-03-24:
There are a number of examples in the "macros" subdirectory. Just look for one that specifies the write_mask. If this is the same size as the word size, it is ommitted. This is the default.

cinder023, 2025-03-24:
> There are a number of examples in the "macros" subdirectory. Just look for one that specifies the write_mask. If this is the same size as the word size, it is ommitted. This is the default.

Thank you!i have got the configuration of writemask.