VLSIDA/OpenRAM #267: Understanding OpenRAM's SPICE Simulation Mechanism and Device Model Configuration Conflict

TemLos, 2025-03-26:
**Describe the bug**
A clear and concise description of what the bug is.
I would like to know the principle of SPICE simulation in OpenRAM, such as what tools are used. I have a device model and wonder if I can configure OpenRAM or modify part of its code to use this model for simulation. I ask this because in tech.py, there is a device model configuration item alongside on-current and saturation voltage items, which seem conflicting.

**Version**
Which commit are you using?
1.2.48

**To Reproduce**
What did you do to demonstrate the bug?
Please include your configuration file used.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Logs**
If applicable, add logs or output to help explain your problem.

**Additional context**
Add any other context about the problem here.

mguthaus, 2025-03-26:
OpenRAM has both an analytical and simulation based characterization. The analytical method estimates things without simulation but will be less accurate but faster. It also needs to be calibrated which it is not in sky130, so those numbers will be inaccurate.

closed 2025-03-26 by mguthaus

TemLos, 2025-04-01:
> OpenRAM has both an analytical and simulation based characterization. The analytical method estimates things without simulation but will be less accurate but faster. It also needs to be calibrated which it is not in sky130, so those numbers will be inaccurate.

Could you please provide some guidance on how to determine whether to use the simulation method? Is it a default option that will be automatically selected, or do we need to make a specific choice to enable it?

mguthaus, 2025-04-01:
Hi,

It is described here in the documentation: 

https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/characterization.md

Best regards

TemLos, 2025-04-02:
> Hi,
> 
> It is described here in the documentation:
> 
> https://github.com/VLSIDA/OpenRAM/blob/stable/docs/source/characterization.md
> 
> Best regards

My apologies. I may not have reviewed the document thoroughly enough. Thank you very much for your assistance.