VLSIDA/OpenRAM #268: Generating SRAM fails at `supply_router`: Couldn't route from (vdd layer=m3 ...) to to (vdd layer=m3 ...)

kammoh, 2025-05-14:
**Describe the bug**

Generating an SRAM using sky130 with the following config:

```python
num_words = 256
word_size = 96
write_size = 96
num_rw_ports = 2
```

After running for about ~30 minutes, I get this error:

```
ERROR: file supply_router.py: line 84: Couldn't route from (vdd layer=m3 ll=v[214.355,-55.435] ur=v[214.845,-54.945]) to (vdd layer=m3 ll=v[214.995,-44.165] ur=v[215.485,-43.675])
```

Same issue with:

```python
num_rw_ports = 1
num_r_ports = 1
```

Same issue using different Python versions.

**Version**
Version 1.2.48 (stable branch commit b6a6f12642df6b84facc24a77f9a6f67a0d62dab)

**To Reproduce**
Please see above.

**Expected behavior**
SRAM generated?

**Logs**
For 1rw1r config:

```
[2025-05-13 20:14:47.056401] RW ports: 1
R-only ports: 1
W-only ports: 0
[2025-05-13 20:14:47.056482] DRC/LVS/PEX is only run on the top-level design to save run-time (inline_lvsdrc=True to do inline checking).
[2025-05-13 20:14:47.056561] Characterization is disabled (using analytical delay models) (analytical_delay=False to simulate).
[2025-05-13 20:14:47.056642] Only generating nominal corner timing.
[2025-05-13 20:14:47.056724] Words per row: None
[2025-05-13 20:14:47.056815] Output files are: 
...
[2025-05-13 20:14:56.045561] ** Submodules: 8.9 seconds
[2025-05-13 20:14:56.183615] ** Placement: 0.1 seconds
[2025-05-13 20:46:04.762107] ERROR: file supply_router.py: line 84: Couldn't route from (vdd layer=m3 ll=v[214.355,-55.435] ur=v[214.845,-54.945]) to (vdd layer=m3 ll=v[214.995,-44.165] ur=v[215.485,-43.675]).

```

**Additional context**
Add any other context about the problem here.

mguthaus, 2025-05-14:
It's an odd, and very large, word size. I'd consider banking or other optimization. 

Often this type of thing is just caused by a pin access problem, but you would need to identify it in the temp output that is saved. The top of the run should tell you what the GDS file name is to investigate.

Sorry, but we don't have resources to fix bugs, but would welcome any improvements as a PR>

ManzoniGiuseppe, 2025-12-09:
In case anyone wants to try fix it, a similar error also happens with freepdk45, and with the following power-of-two word_size:
- 128 (it fails with num_words = 512, but succeeds with 256)
- 256 (it fails with num_words = 64,128, but succeeds with 16)
- 512 (it fails with num_words = 512) 

For example, this configuration:

```
tech_name = "freepdk45"
word_size = 128
num_words = 512
write_size = 8
num_rw_ports = 0
num_r_ports = 1
num_w_ports = 1
```

Fails with:

```
** Submodules: 1417.4 seconds
** Placement: 0.2 seconds
ERROR: file signal_escape_router.py: line 70: Couldn't route from (addr1[8] layer=m3 ll=v[359.485,-3.61] ur=v[359.62,-3.475]) to (addr1[8] layer=m4 ll=v[359.4825,-15.52] ur=v[359.6225,-15.38]).
```

I can use smaller SRAMs in parallel, so it's not the end of the world, but it's not as efficient as a single bigger SRAM.

mguthaus, 2026-10-13:
Still open, but worth retesting against `dev` first: the supply router has changed since you filed this. #314 and #317 both landed on `dev` and touch exactly this path — #317 replaced recursion in the router's bbox tree, which previously blew Python's recursion limit on large macros, and #314 fixed a graph bug where `remove_all_neighbors()` skipped every second neighbor, leaving edges into blocked positions. Neither was aimed at a `Couldn't route` failure specifically, but both change what the router sees.

If it still fails on `dev`, the routing code is in `compiler/router/`, and running with `-v -v` gives the per-net routing detail.

If you would like to take it, a PR against `dev` would be very welcome and we will review and consider it — see [CONTRIBUTING.md](https://github.com/VLSIDA/OpenRAM/blob/stable/CONTRIBUTING.md). Contributions go to the `dev` branch, and the regression suite is run with `cd compiler/tests && make -k -j<N>`.