VLSIDA/OpenRAM #269: Please consider switching from `volare` to `ciel`

hpretl, 2025-06-26:
As support for Volare will not happen in the future, a switch to Ciel (the successor) should be considered.

mguthaus, 2025-06-26:
Hi, we would welcome a PR as we don't have anyone actively maintaining this right now.

hpretl, 2025-06-26:
Cool, here you go: https://github.com/VLSIDA/OpenRAM/pull/270

mguthaus, 2026-10-13:
Already done — closing.

`stable` no longer references `volare` anywhere; the PDK install goes through `ciel`. `requirements.txt` pins `ciel>=2.0.0`, the `Makefile` uses it for the sky130 and gf180mcu PDK targets, and `basic_setup.md` documents `make sky130-pdk` as "This will use ciel to get the PDK." The Nix devShell also provides `pdk-ciel`.

Thanks for raising it.

closed 2026-10-13 by mguthaus