VLSIDA/OpenRAM #271: "Structure name not found at GDS file position" error when using SRAM inside caravel user project

almazmumi, 2025-08-11:
**Describe the bug**  
I created a small project with one sky130_sram_1kbyte_1rw1r_8x1024_8 SRAM in the user project wrapper to figure out how to connect it to the Caravel user project. Caravel now uses Librelane instead of OpenLane. In the final stage, GDSII Stream Out (Magic), I encountered an error:  

> Copying output for cell sky130_fd_bd_sram__openram_dff_1 from $PDKPATH/libs.ref/sky130_sram_macros/gds/sky130_sram_1kbyte_1rw1r_8x1024_8.gds 
> Calma output error: Structure name not found at GDS file position 125868 
> Calma output error: Can't write cell from vendor GDS. Using magic's internal definition

[image]

This error appears for several cells, including sky130_fd_bd_sram__openram_dff_1, sky130_fd_bd_sram__openram_dp_cell_cap_row_1, sky130_fd_bd_sram__openram_dp_cell_replica_1, sky130_fd_bd_sram__openram_dp_cell_dummy_1, sky130_fd_bd_sram__openram_dp_cell_cap_col_1, sky130_fd_bd_sram__openram_dp_cell_1, sky130_fd_bd_sram__openram_sense_amp_1, sky130_fd_bd_sram__openram_write_driver_1, sky130_fd_bd_sram__openram_dp_nand2_dec_1, and sky130_fd_bd_sram__openram_dp_nand3_dec_1.  

I tried compiling a new SRAM using the OpenRAM tool instead of the one in the PDK, but the same errors occurred.  

**Version**  
OpenRAM: ea15a814438e799c08b6d030692dac0ba0c6374d  
Sky130-pdk: f70d8ca46961ff92719d8870a18a076370b85f6c  

**To Reproduce**  
Just SRAM instantiation on top module user_project_wrapper and make with the following config.json file. I attached the log file and config.json:  
[config.json](https://github.com/user-attachments/files/21708145/config.json)  
[flow.log](https://github.com/user-attachments/files/21708165/flow.log)  

**Expected Behavior**  
To successfully run the Librelane flow with sram macro.  

**Logs**  
Attached.

almazmumi, 2025-08-11:
When I compile new sram using sram_compiler.py, it passed the  GDSII Stream Out (Magic) stage and I got new error. 

[image]

[image]

almazmumi, 2025-08-11:
Last question. Should I use the one in the pdk or the one I compiled using sram_compiler.py?

mguthaus, 2025-08-11:
I am not sure what version is in the PDK, so I cannot answer that.

mguthaus, 2025-08-11:
The PDK version is likely older...

almazmumi, 2025-08-11:
It is sky130A. This is the skywater version
SKYWATER_COMMIT=f70d8ca46961ff92719d8870a18a076370b85f6c
export OPEN_PDKS_COMMIT_LVS?=6d4d11780c40b20ee63cc98e645307a9bf2b2ab8
export OPEN_PDKS_COMMIT?=0fe599b2afb6708d281543108caf8310912f54af

mguthaus, 2025-08-11:
I don't maintain that, so I'm not sure. I have my version here:
https://github.com/VLSIDA/sky130_sram_macros/tree/main